heissgetraenk/fcmcua #17: Add to FreeCAD Addon Manager

luzpaz, 2023-01-13:
https://github.com/FreeCAD/FreeCAD-addons#adding-your-workbench-to-the-collection

luzpaz, 2023-01-15:
What are the 3rd party dependencies of this workbench?

heissgetraenk, 2023-01-15:
asyncua, which in turn needs aiofiles

luzpaz, 2023-01-15:
Thanks, are they available via `pip` ? If so what are their official names?

heissgetraenk, 2023-01-15:
pip install asyncua installs both, i believe. 
https://github.com/FreeOpcUa/opcua-asyncio

luzpaz, 2023-01-15:
@chennes will the addon manager autoinstall both if one is the dependency of the other? :point_up_2:

chennes, 2023-01-15:
Yes, the AM just uses pip, so it will follow whatever dependency chain a normal pip install does. You only need to specify your actual dependency, not all the things it depends on.

luzpaz, 2023-01-15:
Are we good to add  'asyncua' (https://github.com/FreeOpcUa/opcua-asyncio) to the https://github.com/FreeCAD/FreeCAD-addons/blob/master/ALLOWED_PYTHON_PACKAGES.txt file ?

chennes, 2023-01-15:
No problems here.

luzpaz, 2023-01-15:
Thanks chennes. Merged in https://github.com/FreeCAD/FreeCAD-addons/pull/248  
Now we need a `package.xml`

closed 2023-01-15 by luzpaz