heissgetraenk/fcmcua #29: FreeCad addon manager install path

heissgetraenk, 2023-01-18:
The workbench uses the directory structure proposed by the [workbench_starterkit](https://github.com/looooo/Workbench-Starterkit)
i.e. the install directory must look like: FreeCAD/Mod/freecad.fcmcua/freecad/fcmcua/init_gui.py etc.

the addon manager currently installs to: FreeCAD/Mod/Fcmcua/freecad.fcmcua/freecad/fcmcua

@chennes What configuration am I missing to install to path as described above?

chennes, 2023-01-18:
You have one directory too many in your repo here: in the advice you see there "freecad.fcmcua" is the name of your GitHub repo (technically, it's the name listed in the .gitmodules file in FreeCAD-Addons). You'll see that basically only loooo names his repos like that, everyone else just gives them the name they want to (e.g. "fcmcua") So you need to remove that directory from your repo here, then you just have freecad/fcmcua/init_gui.py. Make sure to update your package.xml file's icon path as well.

heissgetraenk, 2023-01-18:
That did it, Thanks!

closed 2023-01-18 by heissgetraenk